SensorApps/Sensors2OSC #2: Menu is not visible in Android 4.4.2 on Samsung device

residuum, 2015-04-22:
User reports, that the 3 dots for the menu are not visible on his device. 

It was fixed for him by changing `android:showAsAction="never"` to `android:showAsAction="ifRoom"` in https://github.com/SensorApps/Sensors2OSC/blob/master/app/src/main/res/menu/start_up.xml

This needs to be updated in the repository and probably in Sensors2Pd as well.

residuum, 2015-04-23:
This seems to be by design: http://stackoverflow.com/q/23654854/124983

residuum, 2015-04-24:
User reports back, that menu button on device shows the menu, while the 3 dots for the menu are not visible in the headline.

This is by design and working as intended.

closed 2015-04-24 by residuum

deusanyjunior, 2015-04-28:
I updated the code, but it would be better to know the device model in this case, so we could create an emulator.

residuum, 2015-04-29:
No, this is by design, this is the standard way menus should behave in Android.

deusanyjunior, 2015-04-29:
wow.. so I will roll back with this update..

sorry..

=P
dj