SensorApps/Sensors2OSC #3: App does not build

tkirshboim, 2015-05-19:
Hello,
I tried to build the app and failed at two points:
1. The Common module is referencing a non existing commit so I couldn't init and update that submodule. I managed to fix it but then-
2. I tried to create a release APK but some lint errors got in the way. With Android Studio I could start the app at this point as a debug APK without any problem. To fix this I disabled aborting on errors when checking release build.

I forked this repository and made the fixes on my fork so it can be merged from there:
https://github.com/tkirshboim/Sensors2OSC

residuum, 2015-05-20:
The error was introduced with commit 984f7c92cc708c14eac112976d8d464beaa35d0a

closed 2015-05-20 by deusanyjunior

deusanyjunior, 2015-05-20:
merged!

= )
dj

residuum, 2015-05-20:
The main problem was, that the commit in SensorsApps/Common was not pushed to github.

deusanyjunior, 2015-05-21:
yeah, I am sorry. it is kinda hard to remember to commit in both
repositories all the time.. specially when I change Common while working
inside of Sensors2OSC.

= |
dj

tkirshboim, 2015-12-17:
Hello again - the current master branch is again not building. I fixed it and created a pull request
https://github.com/SensorApps/Sensors2OSC/pull/5

I also added a pd example that uses the pd-vanilla `oscparse` object and created a PR for that:
https://github.com/SensorApps/Sensors2OSC/pull/6

Regarding having the Common module as a submodule - you can have a look at a branch on my fork that uses the Common module as an external gradle dependecy - 
https://github.com/tkirshboim/Sensors2OSC/commit/f1da08a274afc7d6028c6805701f66a167400084

This could prevent issues like a broken submodule reference from happening in the future.

deusanyjunior, 2015-12-19:
> tkirshboim@f1da08a

cool! it is better to change this! thanks a lot!

= )
dj

tkirshboim, 2015-12-19:
Hi @deusanyjunior - I created a pull request for the replacement of the Common submodule with the jitpack dependency so you can have a look and  decide if you'd like to merge it:
https://github.com/SensorApps/Sensors2OSC/pull/7/files

Jitpack is pretty simple to use. You can have a look at their website:
https://jitpack.io/#SensorApps/Common

residuum, 2016-01-13:
The PR is merged.

PR #4 (2015-05-20): Fix submodule reference and disabled aborting release build on lint errors
see / fixes #3