SensorApps/Sensors2OSC #11: Run in Background

residuum, 2017-03-14:
When a user sets "send data" to on, then the app should be able to run in the background.

When the app is sending data , then a notification should be visible in the notification bar.

When the user touches the notification, the app should open.

deflost, 2017-07-23:
hello, we would like to use this super cool little app in teathre, maybe for a dance show to trigger sounds and visuals.
but only when we are able to look the device so the app have to run in backround, it would be possible!!!
it would be great you can bring this feature inside your app.
r.

meta-meta, 2019-05-22:
It would be great if it could run while screen is off as well. https://stackoverflow.com/questions/11978742/how-to-keep-an-activity-running-while-screen-off

ninjaihero, 2021-07-01:
I have let somebody to do it.

ricardoscholz, 2021-07-01:
we have the same idea of @deflost, for usage in theatre. i have made some tests already for DMX lighting control with marine (www.marineframework.org). it works great. but making sure it would keep sending data with the screen off and locked is a must for live usage, actually. thanx for this amazing app! i'm looking forward for this feature! :)

PR #36 (2022-05-20): Background service for sending data
This is an implementation of background services for sending the data.

Will fix #11 and #35 

Currently, NFC tags are not working, because the current method needs an activity.
closed 2022-05-20 by residuum