SensorApps/Sensors2OSC #15: Sensor range and output unclear

mxa, 2017-07-08:
I'm trying to make sense of the sensor values and their specific uses.
https://developer.android.com/reference/android/hardware/Sensor.html has some information and explanations.

It's confusing that the ranges of the sensors are not normalized.

I'd like to contribute with some example Pd/Gem patches, for instance 
https://github.com/mxa/GemVR/blob/master/ReceiveSensors2OSC.pd but it's not ready yet.

residuum, 2017-07-09:
Normalize to what range? 0 - 1? 0 - 127? -1 - 1?

mxa, 2017-07-10:
That's a good question. I'm not 100% it should be normalized, maybe I just can't see the logic behind the current mix of ranges. If it's normalized, I think that 0.0 - 1.0 would make most sense, that would follow the standard in Gem. 0 - 360 would be possible too for rotation. [On stackowerflow](https://stackoverflow.com/questions/27106607/get-rotation-and-display-in-degrees) I'm reading that the current numbers are Radians, so that is probably okay as well, there should just be documentation somewhere so that this is more clear. I'm happy to help there.

residuum, 2017-07-10:
IMHO it is a problem with understanding the different values for rotation:

Rotation around one axis has a range of 0 - 360° (X)
Rotation around the second axis has a range of -180 - 180° (Y)
Rotation around the third axis has a range of -90 - 90° (Z)

Normalization of these values does not make sense, because these are derived from the mathematical properties.

mxa, 2017-07-12:
I think i'm looking for `SensorManager.getRotationMatrix` as in https://stackoverflow.com/questions/15315129/convert-magnetic-field-x-y-z-values-from-device-into-global-reference-frame#15317814

mxa, 2017-07-18:
Feel free to adapt patches ReceiveSensors2OSC.pd and/or Sensors2OSC-help.pd from https://github.com/mxa/GemVR as more verbose helpfiles to Sensors2OSC

mxa, 2017-07-25:
Video which illustrates the difficulty. Turning the phone horizontal will not work. https://vimeo.com/226958567

residuum, 2018-03-02:
Working on it.

residuum, 2018-03-05:
While I am at it, should [inclination](https://developer.android.com/reference/android/hardware/SensorManager.html#getInclination(float[])) be included as well?

residuum, 2018-03-09:
Should be working.

closed 2018-03-29 by residuum