SensorApps/Sensors2OSC #16: NFC data

markeisenberg, 2017-07-11:
Is it possible to differentiate between NFC tags with encoded text or NFC serial number? when I check for the OSC message in an OSC monitor, I get just that it is an NFC and some float value.

deusanyjunior, 2017-07-11:
the values we send are defined in this file: https://github.com/SensorApps/Common/blob/master/src/main/java/org/sensors2/common/dispatch/Measurement.java#L31

this is the main value: msg.getRecords()[0].getId()

in this case, the id is sent after the /nfc. are you getting similar IDs for different sensors?!

markeisenberg, 2017-07-11:
I currently have a set of four NTAG213 sticker tags and also one separate card with NFC which has a different ID. The four sticker tags can be written to and read separately, but return the same IDs (large floats) when checking using various OSC monitors. The four IDs are returned as `/nfc f 3.60xxxe+16` - i.e. the same value. I am using MIOS Studio to monitor the OSC

deusanyjunior, 2017-07-11:
for which options/values your four NFC cards are different?! if all of them have the same ID this result is expected, no?!

markeisenberg, 2017-07-11:
Yes, it is expected. I just tested another one of the cards. The ID seems to correspond to the brand of the NFC tag. So I technically could solve my problem by getting four different NFC tags. I was just wondering if it was possible to extract data that was written to the tags using and NFC writer.

deusanyjunior, 2017-07-11:
let me know the data you want and I can try to send to you.
here you have the Android documentation:
https://developer.android.com/reference/android/nfc/NdefMessage.html

markeisenberg, 2017-07-11:
I'm looking for a unique identifier for each tag. The serial number might be enough for my case, but the ability to extract a pre-programmed text string would be nice. I'm not sure if this data is standard to acquire in all cases. 

> the values we send are defined in this file: https://github.com/SensorApps/Common/blob/master/src/main/java/org/sensors2/common/dispatch/Measurement.java#L31

Is it simply possible for me to compile the source after changing the number on line #36?

`this.values[0] = (float) getDec(msg.getRecords()[0].getId());`

Thanks for the feedback 👍

deusanyjunior, 2017-07-11:
you need to setup your local project to use your version of Common module and change it the way you want. 

in the class Measurement you can also create a private type **string stringValues** and use textual values. in this case you will need to change the method that sends the values through OSC:
https://github.com/SensorApps/Sensors2OSC/blob/master/app/src/main/java/org/sensors2/osc/dispatch/OscDispatcher.java#L32 
and #L43

hope you'll get it done easily, or else the docs are not good enough!!

markeisenberg, 2017-07-11:
Thanks for the info. I'll try to work at it, might be a while. By the way, out of curiosity, why is the default data format set as just `/nfc` and the id number? Was there a use case in mind or was it just an interest in sending one NFC signal?

deusanyjunior, 2017-07-11:
some user asked for /nfc and I implemented this way as I have many cards and they differ only by their ID. I never had cards with the same ID, so I don't know what happens in this situation.

markeisenberg, 2017-07-12:
Hello again. I actually ended up finding a solution with a different approach using Tasker, the Pushbullet API and JSON. I might return to this option if I still encounter problems. Cheers!

closed 2017-07-12 by markeisenberg