SensorApps/Sensors2OSC #17: All sensonrs listed twice

mxa, 2017-07-12:
It appears as all sensors are listed twice with Sensors2OSC 0.2.0. See screenshots and note the faint scrollbar position on the right.

[image]
[image]

residuum, 2017-07-13:
Are you sure? It's the list of sensors that can be scrolled while the main button stays fixed on top.

mxa, 2017-07-13:
Here is the complete list from my phone (Sony Xperia Z3+)

nfc
acceleration
magneticfield
magneticfielduncalibrated
gyroscope
gyroscopeuncalibrated
proximity
light
pressure
accelerometer
magneticfield
magneticfielduncalibrated
gyroscope
gyroscopeuncalibrated
proximity
light
pressure
gravity
linearacceleration
rotationvector
georotationvector
orientation
tiltdetector
gravity
linearacceleration
rotationvector
georotationvector
orientation
stepcounter
stepdetector
significantmotion
gamerotationvector
stepcounter
stepdetector
gamerotationvector

residuum, 2017-07-14:
OK, I have found this and will try to fix that:

> The list of all sensors present on the device is reported by the HAL implementation.
> - There can be several sensors of the same type. For example, two proximity sensors or two accelerometers.
> - The vast majority of applications request only a single sensor of a given type. For example, an application requesting the default accelerometer will get the first accelerometer in the list.
> - Sensors are often defined by wake-up and non-wake-up pairs, both sensors sharing the same type, but differing by their wake-up characteristic.

https://source.android.com/devices/sensors/

residuum, 2018-03-09:
I have no device with more than one sensor of a type, but it should be fixed now. Can you verify it?

mxa, 2018-03-09:
I will test this as soon as an update propagates through f-droid, ok?

residuum, 2018-03-09:
I'd rather have you test it before, as I am unable to test the fix. I will send you a test build.

mxa, 2018-03-10:
Can confirm that sensors are now listed only once.

closed 2018-03-10 by residuum