SensorApps/Sensors2OSC #21: MQTT Transport?

ccoenen, 2018-05-19:
I'm currently working on a project involving an MQTT broker. I am looking into modifying this app, to also send data via mqtt. Do you think it's possibly? From my point of view, it's already doing 95% of all the neccessary things.

I tried understanding the code, and I guess I'd have to (more or less) implement this directory over again, but with mqtt: https://github.com/SensorApps/Sensors2OSC/tree/master/app/src/main/java/org/sensors2/osc/dispatch

What I didn't quite understand, is, where these classes can be found. I didn't find anything with the same name or even the `common` package.
https://github.com/SensorApps/Sensors2OSC/blob/22a3942c0c65befa11e0b297c59956d4d66635d1/app/src/main/java/org/sensors2/osc/activities/StartUpActivity.java#L37-L42

residuum, 2018-05-19:
It would be a good idea to use this app as a template for an additional app for MQTT (Maybe Sensors2MQTT?).

The classes are in a different project: https://github.com/SensorApps/Common/tree/master/src/main/java/org/sensors2/common

ccoenen, 2018-05-20:
Thanks to your help I was able to whip up a _VERY_ basic demo in no time at all :-). It currently can't do very much, and it will only send the first element of a float array, but It connects to a server and sends it successfully!

ccoenen, 2018-05-20:
Here's my (very hackish) MQTT-Fork: https://github.com/ccoenen/Sensors2OSC

With a slight alteration of `DataDispatcher`, I wouldn't have to touch `MultiTouchFragment`:
https://github.com/ccoenen/Sensors2OSC/blob/245e41421029772e5360b3ba493f431895867c37/app/src/main/java/org/sensors2/osc/fragments/MultiTouchFragment.java#L36

If `DataDispatcher` contained `addSensorConfiguration`, the typecast wouldn't be needed:

https://github.com/SensorApps/Sensors2OSC/blob/22a3942c0c65befa11e0b297c59956d4d66635d1/app/src/main/java/org/sensors2/osc/fragments/MultiTouchFragment.java#L31

residuum, 2018-05-21:
Possibly, yes. Sorry, but I have no time to really have a look at the code at the moment.