SensorApps/Sensors2OSC #31: Option to send message in bundle

PanderMusubi, 2019-09-06:
Please offer an option (probably by default off) which sends the OSC message inside an OSC bundle.

Some software on the receiving side require a message to be in a bundle and do not act on unbundled OSC messages. A single message in a single bundle would fix that.

residuum, 2022-05-24:
I am unsure, if the implementation is working. On the one hand, this is putting the values in a bundle, on the other hand, none of my tests produce a different output on the receiver side.

residuum, 2022-05-31:
I know, but does it work with a build from master branch?

> It is some software we cannot change that requires a single message in a single bundle... I know, not optimal.

I have understood that, but I am unsure, if the implementation in master branch is working, both the Pd and Python examples do not detect a difference. Could you test it? If you cannot build it yourself, let me know, and I will send you an APK.

PanderMusubi, 2022-05-31:
At the moment not working with that part of the software, so I will see when a next version is there when I work on it again. To test it, I need to set up a lot of hardware which costs more than one day.

residuum, 2022-06-24:
Release with bundling is out, I have tested the OSC messages.

closed 2022-06-24 by residuum