SensorApps/Sensors2OSC #40: The X axis rotation(orientation2) data will suddenly jitter

RayInst, 2022-10-05:
The X axis rotation data will suddenly jitter when the phone is stationary.
see the green line in the picture
[image]

residuum, 2022-10-05:
For investigating the issue, could you please provide the following information:

What is your device name and Android version? 
Does the value jitter all the time or only at the start of sending the data?
Do the gravity and magnetic field values jitter as well?

I suspect, that there may be two different things at work here. 

From the documentation `TYPE_ORIENTATION This constant was deprecated in API level 8. use SensorManager.getOrientation() instead.`

Sensors2OSC reads all available sensors from the device, and if no native orientation sensor is available, then it uses getOrientation(), as specified in the documentation. For that, it needs both gravity and magnetic field values to compute the rotation matrix.

Maybe your device reports an orientation sensor, and Sensors2OSC uses that values, and the other app uses getOrientation() to compute the values, or the other way round. Therefore I need the information above to pin the issue.

residuum, 2022-10-05:
I have just tried it with my installation, and it seems that all values are jittering, when I set the sensor rate to game or fastest. What is the other app that you compare Sensors2OSC to, and what sensor rate does this app use?

RayInst, 2022-10-06:
This happens with all thThis happens with all the rate modes I've tested. Other apps I use are Zig Sim, HyperIMU..e rate modes I've tested. Other apps I use are Zig Sim, HyperIMU..

residuum, 2022-10-07:
I have contacted the author of HyperIMU to get some information on how he is getting the values without jittering.

closed 2022-10-12 by RayInst

residuum, 2022-10-12:
Hmm.